DalFSAE/vehicle-control-unit #5: Validate functionality of the BSPD

AshtonDudley, 2025-05-05:
The BSPD did not work reliably, and would cause the car to fail during the brakes test. This was fixed by greatly increasing the brake pressure and current signal thresholds.

AshtonDudley, 2025-05-07:
To do this, we would need to open the accumulator and pass a wire through the current sensor.

AshtonDudley, 2025-05-09:
The fault signal of the BSPD is sent to the VCU, but it is not used. It would be useful to log and display the BSPD status on the CAN bus, so that it can be used with the dash

AshtonDudley, 2025-09-03:
We should test the BSPD, and document the results here. Low priority right now, but should be considered when we start working on a new BSPD

Lorne-Mac, 2025-12-30:
Test Conditions:
1. BPS and CUR within operating ranges
    - Both between ~0.5V-2.5V
    - Either one above threshold (~2.5V in our sim) independently
2. Fault conditions
    - Both above threshold value
    - Wire faults
        - Either one falls below ~0.5V
        - Either one rises above ~4.5V
        - Both fall below ~0.5V
        - Both rise above ~4.5V

Truth Table Format:
Should look like table below
Condition on BPS | Condition on Cur | Output |
 -------------------| ------------------- | -------- |
0.5V-2.5V             | 0.5V-2.5V              | High     |
2.5V-4.5V             | 0.5V-2.5V              | High     |
0.5V-2.5V             | 2.5V-4.5V              | High     |
2.5V-4.5V             | 2.5V-4.5V              | Low      |
0V-0.5V                | 0.5V-2.5V              | Low      |
0V-0.5V                | 2.5V-4.5V              | Low      |
0V-0.5V                | 4.5V-5.0V              | Low      |
0.5V-2.5V             | 0V-0.5V                 | Low      |
2.5V-4.5V             | 0V-0.5V                 | Low      |
4.5V-5.0V             | 0V-0.5V                 | Low      |
0V-0.5V                | 0V-0.5V                 | Low      |
4.5V-5V                | 4.5V-5V                 | Low      |