DalFSAE/vehicle-control-unit #16: Draft VCU software architecture

AshtonDudley, 2025-08-22:


DaniyalV1, 2025-11-04:
@Lorne-Mac, are developint VCU software from scratch or just documenting the logic behind it?

If its just documentation, where can I find the details about the code?

If we are making the code, could you please give me some detail about what is the VCU software used for (features etc.) and how you would like us to develop it?

reubenchapple, 2025-11-05:
Update: Haven't started this yet, plan to begin over reading week

AshtonDudley, 2026-05-16:
@NicoLebel, let's use this to track the progress of the documents. I updated the body to include the docs we made today

AshtonDudley, 2026-05-16:
Totally forgot that I started making docs for some of these subsystems on a dev branch. I'm going to start compiling info from my personal notes and other sources. 

[image]